rad1o/hardware #124: Position of jack connector

loelkes, 2015-07-13:
Hi,
der Kopfhöhreranschluss (3,5mm Klinke?) ist auf dem Bild auf einer anderen Stelle als auf dem EAGLE-Plan. Ich denke mal, dass das BIld noch ein Beta-Board war oder?
Ich baue nämlich gerade 3D CAD Dateien. Ich gehe davon aus, dass das EAGLE-File stimmt.
[image]

schneider42, 2015-07-13:
Yes that is correct. We moved the headset jack a bit on the final board. I will close the issue as soon as the final pictures of the design are available online.

loelkes, 2015-07-13:
OK.
Do you have dimensions in z-direction (approximate height of of the parts and batteries?) or did i miss them somewhere?

closed 2015-07-13 by loelkes