rad1o/hardware #125: Case design

loelkes, 2015-07-17:
Hallo,
ich habe mal ein Grundelement für eine Gehäuse designed und wollte testen ob das geht. Es soll verschiedene Versionen geben: Bumper (damit es gut liegt) und dann noch ein paar mehr oder weniger geöffnete Versionen.
Ich könnte euch eins mit der Post schicken oder habt ihr ein neues PCB, dass ich zum testen haben könnte?

Frage: ist es ein 1,6mm PCB?

Der Plan ist schon ein paar viele Cases vor dem Camp zu drucken damit dann vor Ort genug für alle gibt.

[image]

morawekj, 2015-07-17:
Hallo,

so wie es aussieht kommen die neuen Muster am Mittwoch/Donnerstag an, gerne testen wir, ob es passt

in etwa 1.6, müsste mal in den Wagenaufbau schauen, muss mal Schneider anhauen.

Gruß Jan

loelkes, 2015-07-17:
OK, ich schicke euch am Montag/Dienstag ein paar Samples. Magst du mir eine Adresse an obelix@entropia.de schreiben?
Grüße
Obelix

schneider42, 2015-07-17:
Ja, ist ein 1.6 mm Lagenaufbau

closed 2015-12-21 by schneider42